tum-ei-eda/seal5 #116: Tracking Issue: Automated Test/Support Coverage Reporting

PhilippvK, 2024-10-20:
As discussed on Friday, we need an (automated) flow to track and report which kinds of instructions are currently being supported by Seal5. As the project is quite ambitious, I came up with a number of sub-tasks which can be implemented on individually.

**Tasks**

1. Properties of Instructions
- [ ] Define relevant instruction properties
- [ ] Extract instruction properties from metamodel
- [ ] Write instruction properties to file

2. Status/Support of Instructions in Seal5 Flow
- [ ] Extend pass metrics to emit names of failing/skipped instructions 
- [ ] Write status/supported patches to file

3. Tracking LLVM Test Results
- [ ] Map test files to instructions
- [ ] Parse test results
- [ ] Store test results in metamodel
- [ ] Write test results to file

4. Analysis and Grouping
- [ ] Merge artifacts (properties, status, tests) in single table
- [ ] Define useful filters/hierachy/…
- [ ] Automatically combine similar instructions (same properties) into groups
- [ ] Aggregate status/tests per group

5. Generation of Reports
- [ ] Propose visualization methods (static/dynamic, table/tree/grid?)
- [ ] Pick suitable visualization tool
- [ ] Generate Reports
- [ ] Publish reports

6. Automation (CI/CD)
- [ ] Define triggers (weekly/push/tag/…)
- [ ] Add workflows to refresh/release reports
- [ ] GitHub Pages integration?

7. Miscellaneous 
- [ ] Allow combining reports from multiple Seal5 runs/flows
- [ ] Increase number of available tests
- [ ] Increase number of example instructions/extensions

PhilippvK, 2024-10-20:
@thomasgoodfellow @dlr-jas @kebi-be @AndHager Feel free to add any comments or assign any tasks you would like to work on.

Tomorrow I can share some more notes!

PhilippvK, 2024-10-25:
I am making good progress. 

@thomasgoodfellow @kebi-be @dlr-jas 
Using the example CSV snippes I have posted in #117, #120 & #123 you should be able to get started with Task 4. & 5.

I will start working on some CI/CD changes to generate the CSVs automatically and look into adding more test cases!

PhilippvK, 2024-11-01:
@kebi-be @dlr-jas @thomasgoodfellow @AndHager 

I made quite good progress this week:
- Fixed broken Core-V Demo
- Fixed broken unit tests
- Implemented more unit tests
- Generate raw data artifacts in reports
- Write data as markdown to Workflow Summary 

See here for the recent run: https://github.com/tum-ei-eda/seal5/actions/runs/11630150544

Based on this, we you can always get the latest csv files by downloading the `demo-*-reports` artifact. Since the data  is already available, it should be easy to do the combination/analysis and generation of the final reports in a CI job.

Further, I added more properties which should help grouping/filtering the instructions:
- `enc_size`: 16/32/48bit 
- `is_compressed`: enc_size == 16
- `op_code`: (OP-IMM, LOAD, custom-0,…)
- `enc_format`: (r-type, i-type, …)
- `{uses,defs}_custom_reg`: in case non-GPR registers are used, we can not generate patterns
- `has_imm_leaf` (not pushed yet): If the behavior contains operations on immediate inputs which can be constant folded, the generated pattern will never match. This detection should let us skip the pattern-gen stage for incompatible instructions.

kebi-be, 2024-11-05:
Currently using the examples of the OpenASIP_base for Task4 and Task5. This concept will be used for analysing the data in the CI as mentioned by @PhilippvK

kebi-be, 2024-11-07:
@PhilippvK: There is an error in the "test_results.cvs"data from the  https://github.com/tum-ei-eda/seal5/actions/runs/11630150544 which results to error when merging the files.  Used the corev demo reports.
Columns: **model_name**,set,xlen,instr,test_kind,test_file,test_fmt,result,**model** 
Fix: rename **model_name** to **model** and delete redundant column "**model**" at the end

PhilippvK, 2024-11-07:
I will take care of this today, thanks!

PhilippvK, 2024-11-11:
> @PhilippvK: There is an error in the "test_results.cvs"data from the https://github.com/tum-ei-eda/seal5/actions/runs/11630150544 which results to error when merging the files. Used the corev demo reports.
Columns: model_name,set,xlen,instr,test_kind,test_file,test_fmt,result,model
Fix: rename model_name to model and delete redundant column "model" at the end

Fixed!

PhilippvK, 2025-01-09:
I will close this as all remaining steps can be discusses in the related separate issues.

Thank you for your help @kebi-be!

closed 2025-01-09 by PhilippvK