tum-ei-eda/seal5 #136: Merge artifacts (properties, status, tests) in single table

kebi-be, 2024-10-30:


kebi-be, 2024-11-05:
Used Pandas to merge the tables.

model_x,set_x,xlen_x,instr,stage,pass,status,model_y,set_y,xlen_y,is_rv32,is_rv64,num_operands,num_inputs,num_outputs,num_inouts,num_regs,num_gprs,num_imms,imm_types,is_multi_in,is_single_in,is_multi_out,is_single_out,is_mimo,is_miso,is_siso,is_simo,has_side_effects,may_load,may_store,is_terminator,is_branch,uses_custom_reg,defs_custom_reg,model,set,xlen,test_kind,test_file,test_fmt,result
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,transform,transform_passes.simplify_trivial_slices,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.c,c,FAIL
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,generate,generate_passes.pattern_gen.behav_to_pat,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.c,c,FAIL
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,generate,generate_passes.pattern_gen.behav_to_llvmir,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.c,c,FAIL
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,transform,transform_passes.explicit_truncations,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.c,c,FAIL
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,generate,generate_passes.riscv_instr_info,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.c,c,FAIL
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,transform,transform_passes.detect_inouts,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.c,c,FAIL
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,transform,transform_passes.detect_behavior_constraints,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.c,c,FAIL
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,transform,transform_passes.detect_side_effects,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.c,c,FAIL
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,transform,transform_passes.collect_immediate_operands,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.c,c,FAIL
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,transform,transform_passes.collect_register_operands,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.c,c,FAIL
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,generate,generate_passes.pattern_gen.behav_to_llvmir,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.ll,ll,PASS
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,generate,generate_passes.riscv_instr_info,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.ll,ll,PASS
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,transform,transform_passes.detect_inouts,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.ll,ll,PASS
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,transform,transform_passes.detect_side_effects,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.ll,ll,PASS
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_MAX,transform,transform_passes.collect_register_operands,success,OpenASIP_base,OpenASIP_base,32,True,False,3,2,1,0,3,3,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,OpenASIP_base,OpenASIP_base,32.0,cg,openasip_base_max.test-cg.ll,ll,PASS
[... cut ...]
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_SELECT,generate,generate_passes.riscv_instr_info,success,OpenASIP_base,OpenASIP_base,32,True,False,4,3,1,0,4,4,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,,,,,,,
OpenASIP_base,OpenASIP_base,32,OpenASIP_base_SELECT,generate,generate_passes.pattern_gen.behav_to_llvmir,success,OpenASIP_base,OpenASIP_base,32,True,False,4,3,1,0,4,4,0,set(),True,False,False,True,False,True,False,False,False,False,False,False,False,False,False,,,,,,,

PhilippvK, 2024-11-05:
@kebi-be The contents are missing here? (I saw them in the mail)

Could you make sure to drop the rendundant `model,set,xlen` cols in the merged df?

How do you handle instructions which do not have tests available? (an inner-join would drop those but we should have information about the non-existent tests in the table)

kebi-be, 2024-11-05:
I deleted the files as I noticed the redundant columns in the df. Instructions without tests will simple have an NaN value.  I will update this. I will add NaN in the empty fields of the non-existing tests.

I first merge the status and properties tables (inner join) and then merge the results to the status_prop_table. Since for the OpenASIP_base_MAX tests results there are  5 different tests and for this instruction there are 10 different passes from the status table. So combining these will results to the multiple rows (5x10). What are your expectations. Otherwise I donot see any other way of combining these.

PhilippvK, 2024-11-06:
I think we have to aggregate the data to display it in a meaningful way. Of course we will loose some information (i.e. names of failing passes/tests) but should get some overall (per instruction, per-class, per-set, total) metrics on how good Seal5 peforms.

In the following comments I will post some ideas on how the aggregations could be done.

PhilippvK, 2024-11-06:
For the Status Data (last row aggregates over a  instructions in the set)

| set           | instr                 | n_success | n_skipped | n_failed | Summary<br>(pass/skip/fail)<br>[%] | Status<br>(good/ok/bad)<br>[%]  |
|---------------|-----------------------|-----------|-----------|----------|------------------------------------|---------------------------------|
| OpenASIP_base | OpenASIP_base_SHL1ADD | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_SHL2ADD | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_ADD     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_SUB     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_EQ      | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_GT      | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_GTU     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_LT      | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_LTU     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_NE      | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_GE      | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_GEU     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_LE      | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_LEU     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_SHL     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_SHR     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_SHRU    | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_AND     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_IOR     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_XOR     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_MIN     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_MAX     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_MINU    | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_MAXU    | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_MUL     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_MULHI   | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_MULHIU  | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_MULHISU | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_DIV     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_DIVU    | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_MAC     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_ROTL    | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_ROTR    | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_MOD     | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
[... cut ...]
| OpenASIP_base | OpenASIP_base_REMU    | 10        | 0         | 0        | 10 / 0 / 0<br>[100%/0%/0%]         | good                            |
| OpenASIP_base | OpenASIP_base_LMBD    | 9         | 1         | 0        | 9 / 1 / 0<br>[90%/10%/0%]          | ok                              |
| OpenASIP_base | OpenASIP_base_SELECT  | 8         | 0         | 2        | 8 / 0 / 2<br>[80%/0%/20%]          | bad                             |
| OpenASIP_base | *                     | 387       | 1         | 2        | 387 / 1 / 2<br>[99.2%/0.3%/0.5%]   | 36 / 1 / 1<br>[92.3%/2.6%/2.6%] |

PhilippvK, 2024-11-06:
Same idea for the test results data (added dummy test data for MAXU):

| set               | instr              | n_pass | n_failed | n_tests | Summary<br>(pass/fail)<br>[%] | Status<br>(good/bad/unknown)     |
|-------------------|--------------------|--------|----------|---------|-------------------------------|----------------------------------|
| OpenASIP_base     | OpenASIP_base_MAX  | 4      | 4        | 8       | 4 / 4 <br>[50%/50%]           | good                             |
| OpenASIP_base     | OpenASIP_base_MAXU | 0      | 6        | 6       | 0 / 6<br>[0%/100%]            | bad                              |
| OpenASIP_base_MUL | OpenASIP_base_MUL  | 0      | 0        | 0       | 0 / 6<br>[-%/-%]              | unknown                          |
| ...               | ...                | ...    | ...      |         | ...                           | ...                              |
| OpenASIP_base     | *                  | 4      | 10       | 14      | 4 / 4 <br>[50%/50%]           | 1 / 1 / 1<br>[33.3%/33.3%/33.3%] |

PhilippvK, 2024-11-06:
I also had the idea to give a metric for the test coverage of each instruction. Here is an example of what I mean:

A Test Coverage of 100% means the following tests (`kind+fmt`) are being executed for the instruction:
- `mc+s`
- `inline-asm+s`
- `cg+c` (if pattern-gen is not skipped)
- `cg+mir` (if pattern-gen is not skipped)
- `cg+ll` (if pattern-gen is not skipped)
- `legalizer+mir` (if legalizer setting exists for instruction)
- `intrin+c` (if intrinsic exists for instruction)
- `builtin+ll` (if intrinsic exists for instruction)

Here is a table with some manually entered coverage data:

| set               | instr              | Coverage<br>(existing/missing/goal)<br>[%] | Status<br>(good/ok/unknown)      |
|-------------------|--------------------|--------------------------------------------|----------------------------------|
| OpenASIP_base     | OpenASIP_base_MAX  | 8 / 0 / 8 <br>[100%]                       | good                             |
| OpenASIP_base     | OpenASIP_base_MAXU | 6 / 2 / 8<br>[75%]                         | ok                               |
| OpenASIP_base_MUL | OpenASIP_base_MUL  | 0 / 8 / 8<br>[0%]                          | unknown                          |
| ...               | ...                | ...                                        | ...                              |
| OpenASIP_base     | *                  | 14 / 10 / 24 <br>[58%]                     | 1 / 1 / 1<br>[33.3%/33.3%/33.3%] |

PhilippvK, 2024-11-06:
@kebi-be If you think the coverage information would be nice-to-have, then I will add it to the Tracking issue #116 and implement a report backend which can emit this data.

PhilippvK, 2024-11-11:
@kebi-be As requested, I have added a `--compact` mode for the test results and status (passes) reports. The resulting tables only have a single row per instruction.

All artifacts can be found here: 

**Minimal Example:**

### Passes

| model   | set      |   xlen | instr        |   n_success |   n_skipped |   n_failed |   n_total | status              |
|:--------|:---------|-------:|:-------------|------------:|------------:|-----------:|----------:|:--------------------|
| Example | XExample |     32 | CV_SUBINCACC |          12 |           0 |          0 |        12 | good :green_circle: |

### Test Results

| model   | set      |   xlen | instr        |   n_pass |   n_fail |   n_tests | state               |
|:--------|:---------|-------:|:-------------|---------:|---------:|----------:|:--------------------|
| Example | XExample |     32 | CV_SUBINCACC |        7 |        0 |         7 | good :green_circle: |

PhilippvK, 2024-11-11:
An initial test coverage analysis is also available using the new `--coverage` flag of the `seal5.backends.report.test_results.writer` backend.

**Example Output:**

`test_coverage_compact.md`
| model          | set            |   xlen | instr                 |   n_exists |   n_required |   n_optional |   n_extra |   n_required_exists |   n_optional_exists | coverage            |
|:---------------|:---------------|-------:|:----------------------|-----------:|-------------:|-------------:|----------:|--------------------:|--------------------:|:--------------------|
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_ADD     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_AND     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_DIV     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_DIVU    |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_EQ      |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_GE      |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_GEU     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_GT      |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_GTU     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_IOR     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_LE      |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_LEU     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_LT      |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_LTU     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MAC     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MAX     |          9 |            4 |            5 |         1 |                   4 |                   5 | good :green_circle: |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MAXU    |          7 |            4 |            5 |         0 |                   4 |                   3 | ok :orange_circle:  |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MIN     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MINU    |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MOD     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MODU    |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MUL     |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MULHI   |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MULHISU |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
| OpenASIP_base  | OpenASIP_base  |     32 | OpenASIP_base_MULHIU  |          0 |            4 |            5 |         0 |                   0 |                   0 | bad :red_circle:    |
[... cut ...]
| OpenASIP_paper | OpenASIP_paper |     32 | SEAL5_ROTR32          | cg          | mir        | True       | False   | False    | bad :red_circle:    |
| OpenASIP_paper | OpenASIP_paper |     32 | SEAL5_ROTR32          | legalizer   | mir        | False      | False   | False    | ok :orange_circle:  |
| OpenASIP_paper | OpenASIP_paper |     32 | SEAL5_ROTR32          | intrin      | ll         | False      | False   | False    | ok :orange_circle:  |
| OpenASIP_paper | OpenASIP_paper |     32 | SEAL5_ROTR32          | builtin     | c          | False      | False   | False    | ok :orange_circle:  |

closed 2025-01-09 by PhilippvK