tum-ei-eda/seal5 #155: Proposal for refactored Intrinsics configuration

PhilippvK, 2024-12-30:
Proposed changes:
- Move intrinsics from `settings.intrinsics` to `settings.extensions.{ext_name}.intrinsics` (easier lookup)
- Add signedness information to arguments and return type
- Allow specifying allowed value ranges (i.e. [0, 31] or [-16, 15] for 5-bit operand exposed via `uint8_t` argument)
- Use `c_type` instead of `ir_type` as the default format to define argument types (includes sign information)
- Allow specifying defaults for intrinsics/builtins (on per-extension level)
- Allow just generating an LLVM intrinsic without generating the respective Clang builtin, i.e. via `builtin: null`
- Define/override the builtin/intrinsic prefix per-instruction or per-extension, i.e. via `builtin.prefix: "__builtin_riscv_{arch}"` or `intrinsic.prefix: "llvm.int.riscv.{arch}"`
- Support intrinsic properties (`IntrNoMem`, `IntrWillReturn`, ...) and builtin attributes (`NoThrow`, `Const`, ...). Either explicitly or automatically inferred by Seal5 passes.

This is mostly not implemented in Seal5 yet, so I am happy about any comment.

In the following, I will provide examples to compare the new and old configuration.

**Example 1 (before)**:
```yaml
---
extensions:
  XCoreVAlu:
    # feature: XCValu
    # arch: xcvalu
    version: "1.0"
    experimental: false
    vendor: true
intrinsics:
  # See: https://github.com/openhwgroup/core-v-sw/blob/master/specifications/corev-builtin-spec.md#listing-of-multiply-accumulate-builtins-xcvmac
  intrinsics:
  # unary
  - &ScalarCoreVAluGprIntrinsic
    args:
    - arg_name: rs1
      arg_type: i32
    set_name: XCoreVAlu
    instr_name: SEAL5_CV_ABS
    intrinsic_name: alu_abs
    ret_type: i32
  - <<: *ScalarCoreVAluGprIntrinsic
    instr_name: SEAL5_CV_EXTHS
    intrinsic_name: alu_exths
  - <<: *ScalarCoreVAluGprIntrinsic
    instr_name: SEAL5_CV_EXTHZ
    intrinsic_name: alu_exthz
  - <<: *ScalarCoreVAluGprIntrinsic
    instr_name: SEAL5_CV_EXTBS
    intrinsic_name: alu_extbs
  - <<: *ScalarCoreVAluGprIntrinsic
    instr_name: SEAL5_CV_EXTBZ
    intrinsic_name: alu_extbz
  # binary
  - &ScalarCoreVAluGprGprIntrinsic
    args:
    - arg_name: rs1
      arg_type: i32
    - arg_name: rs2
      arg_type: i32
    set_name: XCoreVAlu
    instr_name: SEAL5_CV_SLET
    intrinsic_name: alu_slet
    ret_type: i32
  - <<: *ScalarCoreVAluGprGprIntrinsic
    instr_name: SEAL5_CV_SLETU
    intrinsic_name: alu_sletu
  - <<: *ScalarCoreVAluGprGprIntrinsic
    instr_name: SEAL5_CV_MIN
    intrinsic_name: alu_min
  - <<: *ScalarCoreVAluGprGprIntrinsic
    instr_name: SEAL5_CV_MINU
    intrinsic_name: alu_minu
  - <<: *ScalarCoreVAluGprGprIntrinsic
    instr_name: SEAL5_CV_MAX
[... 295 lines cut ...]
      - instr_name: SEAL5_CV_ADDUN
        intrinsic: "TODO"
      - instr_name: SEAL5_CV_ADDRN
        intrinsic: "TODO"
      - instr_name: SEAL5_CV_ADDURN
        intrinsic: "TODO"
      - instr_name: SEAL5_CV_SUBN
        intrinsic: "TODO"
      - instr_name: SEAL5_CV_SUBUN
        intrinsic: "TODO"
      - instr_name: SEAL5_CV_SUBRN
        intrinsic: "TODO"
      - instr_name: SEAL5_CV_SUBURN
        intrinsic: "TODO"
```

This the compact syntax it should also be easy to embed the builtin settings (optionally of course) in the CDSL syntax via attributes. Here is an example:

```c
SEAL5_CV_ABS [[builtin="int32_t __builtin_riscv_xcorevalu_alu_abs(int32_t x)"]] {
  encoding: ...
  assembly: ...
  behavior: {...}
}
```

PhilippvK, 2024-12-30:
@thomasgoodfellow @kebi-be @dlr-jas

thomasgoodfellow, 2025-01-06:
Looks ambitiously comprehensive, or comprehensively ambitious :-) Scattershot shallow comments...

- Is the motivation for using c_type over ir_type principally to include the signed-ness info? Seems good anyway since the types are more obvious, maybe more dominant in future LLVM evolution
- Number ranges might also be useful for subsetting - eg like restricting the register numbers in RV32E to [0, 15] (though that's a poor example though since those are also the RV32I instruction encodings so could only work if the toolchain is being built to only support RV32E)
- Custom immediate handling in general is (was? did it get better in LLVM19?) regrettably ugly with the need to generate C++ rather than expressing purely through TableGen
- With some different layers of config possible, e.g. the CoreDSL annotations from #69, prefix overrides here, clear diagnostics will be a blessing for understanding the results 
- As you note the wordiness of the configuration is an existing problem with too much information being duplicated from the CoreDSL types. And some of those anchor patterns like "binary_args_unsigned" might be common/generic enough to be worth promoting to some "standard_intrinsic_patterns.yml" (though